chicagoedt/revo_robot #67: do we need stereo vision to detect sawhorse obstacles?

basheersubei, 2015-01-09:
Run a test with the TiM Laser (at the correct Scipio height) pointing at a sawhorse obstacle at different angles. Rosbag that data. Try moving the laser up and down (use the one not mounted on scipio and attach it to the T-slot with screws adjustable at different heights).

If the laser can consistently detect the sawhorse, then there is no need for stereo cameras. (moving the laser is also an option if it detects the sawhorse when moved up/down)

l0g1x, 2015-01-21:
Tested and confirmed that the SICKTIM551 will be able to pick up a object as thin/thinner then a saw horse leg. Tim has the dimensions of the object (i think it was 5/8ths but ill double check)

basheersubei, 2015-01-22:
so the answer is NO, right?

basheersubei, 2015-02-21:
confirmed no, we'll use the LRF

closed 2015-02-21 by basheersubei