chicagoedt/revo_robot #69: rosbag for Tom - robot_localization

l0g1x, 2015-01-25:
Tom from robot_localization asked if I could give him a rosbag of real world data from our robot.

 - [ ] Imu data
 - [ ] Odometry data
 - [ ] GPS data

Here is the reference to the general issue on his package: https://github.com/cra-ros-pkg/robot_localization/issues/151#issuecomment-71384652

l0g1x, 2015-01-25:
Odometry data will have to be static covariances for now since we haven't modeled any dynamic covariances yet.

l0g1x, 2015-03-04:
Sent Tom a Bag a while ago, but it did not include GPS data since it was super cold outside.

ayrton04, 2015-05-29:
Hey!

Just an update on this: I went ahead and recorded a bag file using our Husky, and I recorded it with an IMU that is mounted in a non-neutral orientation. The data set is in the test directory as test1.bag. What I'd like to do is add more tests to robot_localization that use the same bag file, but tweak different parameters: turning relative mode on and off for each sensor, turning differential mode on and off, using 2D mode, etc. It really comes down to just adding a few new .test launch files and then adding those tests inside of the CMakeLists.txt file. Let me know if you have time and are interested.

-Tom

l0g1x, 2015-05-29:
Hey Tom! I think i should be able to get you this by the end of the weekend, or at least that’s what I'm aiming for. 
Couple questions: 
1) When you say turning a mode on/off, do you mean you want to toggle between the two, **during** a single test? So one second using the differential, and then disable it the next second? Or do you simply want all different type of scenarios all with their respective test?

2) I still have to study on how you currently test the output of tests, but if you could also give me some insight as to how you *numerically* go about testing the outputs of the tests? For example, do you simply play some bag data that you know the sensor readings should tell you you went 10 meters forward, and 2 meters to the right, and then just compare the output of the filter to a hard coded value? If so, how do you go about accuracy? 

If the goal of these tests are just for pure functional testing purposes, then just disregard the second question. 

Thanks!

ayrton04, 2015-06-01:
Hi Krystian,

I have literally zero time constraints on this, so if you are kind enough
to get to it at all, it can be on your own watch.

I want separate tests where we change parameters (i.e., no parameter
changes on the fly). The data set I have includes Husky wheel odometry, IMU
data from an IMU that is mounted in a non-neutral orientation, and GPS
data. For now, I just want to create new tests for the EKF and UKF wherein
we change a single parameter, e.g., one test with differential mode on for
that data set, and one test with it off, then one test with relative mode
on, and one test with it off.

The tests involve the use of a simple test node (already written) that
simply compares the last reported position to a "known" position. I get the
known position by (a) making sure the software is working, and then (b)
running the launch/bag file 30 times in a row and recording the stats on
the final reported position. I wrote a script that automates that process
for you, and it even spits out a nice MATLAB statement for computing the
stats to put into the final test launch file.

Anyway, have a look at the launch and bag files for test1 in the test
directory. You can model everything off of that (in fact, you can really
just copy it completely, and change the parameters in question, and then
add the test to the CMakeLists.txt file).

Thanks a ton! I really appreciate it.

-Tom