chicagoedt/revo_robot #70: Implement robot_upstart into Radxa's

l0g1x, 2015-03-04:
Robot_upstart is a package which allows you to create Ubuntu Start-up processes. We need to set this up to start all the driver node's, just not the nav stack.

http://wiki.ros.org/robot_upstart

basheersubei, 2015-03-04:
sounds easy as hell (and worth it). Just run something like

`$ rosrun robot_upstart install myrobot_bringup/launch/base.launch`

l0g1x, 2015-03-04:
In order to prevent having to keep a sensor permanently assigned to one of the boards, we need to write a script that checks whether /dev/ttyACM0 (we need to udev this so its /dev/roboteq) is available on that board. That way, it doesnt matter what board the sensor is plugged into, the script will determine what background processes to actually execute

Krystian Gebis
University of Illinois at Chicago

basheersubei, 2015-03-04:
yeah, so we can write scripts for these. Hint: they have a Python API for robot_upstart

;)

closed 2015-11-18 by l0g1x