chicagoedt/revo_robot #78: fresh install on scipiotop

basheersubei, 2015-05-25:
we need to be running Indigo and Ubuntu 14...

 - back up any data we need (rosbags or code that's not checked in)
 - install ubuntu
 - install indigo and all dependencies (try the install bash script)

sabinoj, 2015-05-25:
Keep in mind that we have two 64GB hard drives in Scipio, so if you would
like to just back up all of the bags off of the other one and set a fresh
install of Ubuntu 14.04 on it I think that would be the best idea.

          --John Sabino

basheersubei, 2015-05-25:
Bags shouldn't be that big and I have a couple of 16 GB flash drives. Also I have a TB of Dropbox space so I can just plop them there. I just have to get the time to do it

closed 2015-06-11 by l0g1x