chicagoedt/revo_robot #79: convert line_detection output to pointcloud

basheersubei, 2015-05-25:
possible methods:

- [ ] [inverse perspective mapping](http://stackoverflow.com/questions/17241323/inverse-perspective-mapping-when-to-undistort)
- [ ] [the methods in our wiki page](https://github.com/chicagoedt/Software_IGVC/wiki/Computer-Vision-and-ROS#going-from-image-coordinates-pixels-to-world-coordinates-xyz-relative-to-camera)

basheersubei, 2015-05-25:
currently trying with `projectPixelTo3dRay`, which returns a 3dray corresponding to each pixel. Then, we can find the x and y values that intersect with a plane at a certain height.

But first, we have to deal with the fact that the coordinate systems of images are different. Check [here](http://wiki.ros.org/image_pipeline/CameraInfo).

- [ ] create a static transform from camera_optical frame to camera that fixes the difference in frames.

basheersubei, 2015-05-26:
- [ ] create a node that:
    - [ ]  looks up transform from base_link -> camera -> camera_optical
    - [ ] then applies the transform on the rays returned from `projectPixelTo3dRay`
    - [ ] finds intersection of this new ray with the ground plane (z = 0)
    - [ ] fills up a 2d array of vec2f with (x,y) coordinate for every pixel

- [ ] then write a node that:
    - [ ] for every non-zero pixel in the input image, add a point to the pointcloud (with the x,y coordinates looked up from the previously built project and z=0)
   - [ ] publish pointcloud

basheersubei, 2015-05-31:
essentially works. Needs precise testing.

oxoocoffee, 2015-05-31:
Very nice…

basheersubei, 2015-06-15:
will test in Gazebo and real life

closed 2015-06-15 by basheersubei