chicagoedt/revo_robot #81: pixel_coordinate_calc improper exit

l0g1x, 2015-05-29:
After im done running pixel_to_coordinate_calculator.py and then press ctrl+c i get this error log:

    Traceback (most recent call last):
      File "/home/edt/igvc/l0g1x_ws/src/Software_IGVC/line_detection/src/pixel_to_coordinate_calculator.py", line 253, in <module>
        rate.sleep()
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/timer.py", line 99, in sleep
        sleep(self._remaining(curr_time))
      File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/timer.py", line 157, in sleep
        raise rospy.exceptions.ROSInterruptException("ROS shutdown request")
    rospy.exceptions.ROSInterruptException: ROS shutdown request
    shutting down processing monitor...
    ... shutting down processing monitor complete
    done

@basheersubei can you fix this so that the node simply exits on completion? i think thats simplest way to do it

basheersubei, 2015-05-29:
That doesn't seem like an error. It just raises an interrupt exception (which you raised when you hit ctrl+c), and the exception is caught by rospy (that's what the spin() does in rospy) and it shuts down the node. I don't see a problem here.

basheersubei, 2015-06-15:
fixed in f83a732d53b85c0417d214c2636d0a7965ef3ff0

closed 2015-06-15 by basheersubei