chicagoedt/revo_robot #83: Transfer github wiki to EDT wiki

l0g1x, 2015-08-27:
I started working on this yesterday, but I wont be able to do it all myself. Try to follow the same structure as what I currently have. As a matter of a fact, things like Lane detection can have their own page. Type in the URL something like http://wiki.chicagoedt.org/wiki/Software_Engineering/Revo/Lane_Detection and then click create page (you can also just click 'edit' [not 'edit source' as you'll hate your life] and then click create page). Make sure to somehow link it to the root Revo page (http://wiki.chicagoedt.org/wiki/Software_Engineering/Revo)  

I assigned it to @basheersubei as I know this is also a task for myself. @staylo32 can you work on this as well?

l0g1x, 2015-09-23:
Whats the progress on this? @basheersubei do you think we should delete the pages here on the github wiki if this is done?

basheersubei, 2015-09-23:
All the relevant pages are moved (except the home page). Go ahead and
delete the pages (they'll always stay in history since the wiki itself is a
git repo).

basheersubei, 2015-09-26:
I think it's ok to leave the wiki pages here. Let's just stop updating them. (update the main page to link to our chicagoedt wiki)

closed 2015-11-18 by l0g1x