chicagoedt/revo_robot #86: Use Stereo SLAM for wheel odom rotational covariance

l0g1x, 2015-09-28:
Ive been thinking for a while now on how we can estimate (better) the rotational position/velocity done by the robot, and i had a few ideas:
 -  Attach a aruco marker to the top of the robot and have a overhead camera that records the rotation made. We would use the RC remote to do a full 360 degree turn, and then compare the angular position data produced by the wheel odometry, to then be able to specify a more accurate rotational covariance.
 - Have some visual slam that produces some type of odom, and use that in the covariance calculation of wheel odometry. We would only use the rotational component of the visual slam *solely* for wheel odometry rotational covariance calculations. The visual slam odom values would be its own input. We would just use the visual odom as a rotational comparison to what we got on the rotational wheel odometry

Thoughts on this?

l0g1x, 2015-09-28:
I think the second is better as we dont have to 'calibrate' the covariance on different frictional surfaces

staylo32, 2015-09-28:
The visual slam works based on feature detection, correct? And if so, wouldn't that possibly introduce error in odom calculations?

l0g1x, 2015-09-29:
Yes it does, but you can also think of it as using landmark observations on a very small time change level , which will give us a point of reference as to the velocity we are turning at visually. Ill explain more on thursday

closed 2015-11-18 by l0g1x