chicagoedt/revo_robot #89: Broken link for rosbag repository

bergercookie, 2016-09-16:
In the [rosbag-repo page](https://github.com/chicagoedt/revo_robot/wiki/Rosbag-Repo) the Dropbox URL provided seems to be broken.

Can you fix this? Is there an alternative download link?

basheersubei, 2016-09-16:
sorry about that, I fixed the URL

closed 2016-09-16 by basheersubei

l0g1x, 2016-09-16:
@bergercookie would you be able to share the results you get from this dataset in your project? The datasets are recorded camera data for a robotics competition IGVC and our organization would love to learn from whatever you end up doing with the data (if confidentially possible)

bergercookie, 2016-09-16:
@l0g1x Currently I develop a SLAM algorithm that makes use of laser scans and odometry data but support for camera data, 3D Point clouds is on its way. In case I use the datasets I will not hesitate to provide feedback.

l0g1x, 2016-09-16:
@bergercookie awesome thanks! good luck on your project. Feel free to ask us for any help, someone will try to respond when they can